mkopa/speeed #4: Blank screen on Nexus 7 with 6.0.1

brickdigits, 2021-07-24:
Have used Speeed successfully on a Pixel and think what you've achieved is great. However, when attempting to run on a Nexus 7 with 6.0.1 I get the splash screen followed by a blank white screen and that's it. Is a screen resolution compatibility issue?

mkopa, 2021-07-24:
Unfortunately, older smartphones were left without updates, so the Speeed
app doesn't work on them.
What you can do:
1. check/look in the smatphone settings for an update of the WebView and/or
Chrome browsers if you have one.
2. unfortunately you have to try to update the system or use the app on
another smartphone.

brickdigits, 2021-07-25:
Ah, that's frustrating. A friend repaired the Nexus 7 for me specifically so it could be used to run the app, as the app system requirements stated 6.0 and up. Guess I need to find another Pixel to run it on

brickdigits, 2021-07-25:
Ah, that's frustrating. A friend repaired the Nexus 7 for me specifically so it could be used to run the app, as the app system requirements stated 6.0 and up. Guess I need to find another Pixel to run it on

> Unfortunately, older smartphones were left without updates, so the Speeed app doesn't work on them. What you can do: 1. check/look in the smatphone settings for an update of the WebView and/or [...]

closed 2021-08-24 by mkopa