mkopa/speeed #5: Option to show voltage instead of "battery life"

Rombutan, 2021-07-30:
Title.

I want to see plain voltage instead of battery life.

It would be very easy to implement, and I'll even do it myself if you email me the source code. I'm not sure why you dont expose source code, you aren't charging anything or adding advertisements. But there are some other additions I would love to make, even as a long term contributor.

Tonyco2705, 2021-07-30:
yes, battery voltage would be great. also my controller is at 55amps  so the watts aren't correct. so a way to change this would be great

Rombutan, 2021-07-30:
If the author of this project dosent reply I'll make my own, actually open source version. It's just usbserial plus some logic plus a gui. If anyone wants to help...?

mkopa, 2021-07-31:
In the current version, the display is a tool to help configure PAS modes, it is not intended as a replacement for the original display.

The motor controller gives the energy value as a percentage. I don't know the algorithm how it counts it, not sure if it is linear. The factory displays measure voltage and give it in volts (correct me if I am wrong).

I will soon announce the release of a bluetooth adapter and a new version of the Speeed app, which will include a good quality display among other things.

Rombutan, 2021-07-31:
Ok, but can I have your source code to implement it in my own version? I dont want to muck around with bluetooth, there is no need for me, I am happy with a single wire. But I want to write in a faster update rate and a voltage display. Ill characterize the "algorithm" myself.

I am happy to sign an NDA or just give you my "word", and share any advancements I make.

Frankly I would appreciate it if you could declare your intentions as "I want to capitalize off this later" or "I just like helping out" or "I dont know how to make this open source", because I dont know what you are trying to achieve here.

closed 2021-08-24 by mkopa