mkopa/speeed #11: Connection via Bluetooth

deadrabbit87, 2021-11-22:
Hi,

your tool is really awesome! Thanks a lot for your work. 

You wrote, you will provide a connection via Bluetooth. How is the state?

BW Maximilian

Sevy13, 2022-03-02:
Very good job... quand auront nous je Bluetooth même si il faut donner de l'argent
@Sevy13

techjaw, 2022-07-31:
Hello, 
will this work with BAFANG DP E12.CAN? Man, it would be great, I would be happy to pay for such functionality.

closed 2022-11-03 by mkopa

rayholland, 2022-11-13:
bluetooth could be achieved with this cheap uat2ble transceiver module (it's 5v, so it could be plugged directly into the bafang harness):
https://www.go2tech.nl/winkel/componenten/at-09-bluetooth-4-0-ble-module-cc2540-cc2541/

but this requires some code changes from Marcin.
@mkopa ,would you be interested in implementing this?

MartinPicha68, 2023-02-27:
I would also like Bluetooth, can you make your version 1.3.2 available?