mkopa/speeed #17: Is this project abandoned?

devdev86, 2022-09-25:
I would like to see this project continue to progress but it appears to have no activity in about a year - @mkopa  can you confirm if Speeed is abandoned? If so, any chance of a source release to allow others to continue the development? If not, any timeline or plans for bluetooth dongle availability, next version of the application, features, etc.?

closed 2022-11-03 by mkopa

MartinPicha68, 2023-02-27:
I would like to see this project continue to progress but it appears to have no activity in about a year - @mkopa can you confirm if Speeed is abandoned? If so, any chance of a source release to allow others to continue the development? If not, any timeline or plans for bluetooth dongle availability, next version of the application, features, etc.?

mkopa, 2023-04-13:
@devdev86 @MartinPicha68 Apologise. This project will rather be archived. I will decide on it by the middle of the year. Thanks for being there!