mkopa/speeed #24: Not working on android 13

supersixseven, 2023-11-03:
Error message 'This app isn't available on your device because it was made for an older version of android' appears on the play store.

[image]

[image]

3dbluebot, 2023-11-07:
I have the same problem on my 'old' S21 Ultra, it's also Android 13. Can't find workaround unfortunately

DrThrax741, 2023-11-19:
@3dbluebot As a workaround, you can sideload the APK file from a mirror such as apkpure or apkmirror. Download the APK file to your Android device, then open/launch the file with a file explorer app of your choice. The default file explorer that is preinstalled on your device should work fine for this (in my case, My Files). If you do not have a file explorer app, then, Cx File Explorer from the google play store works fine too. If it is your first time side loading with that file explorer app, then a pop-up will prompt you to give the file explorer app permissions to install the app.

https://apkpure.com/speeed-bafang-bbs01-02-bbshd/com.mkopa.speeed/download/1.3.1

I was able to download, install, and launch Speeed through this method on Android 13 (Samsung Z Flip5), but I haven't tested the app's full functionality yet.

3dbluebot, 2023-11-20:
That's great thank you so much for your help.

I'll follow your advice

Kindest regards

Paul

Dweebur2, 2024-01-23:
I was able to load however unable to save profiles.  Not very useful